wenbowen123/BundleTrack #8: Segmentation fault

mkt1412, 2021-11-16:
Hi,

I was trying to run the pretrained model on NOCS with command "python run_nocs.py --scene_id 1 --port 5555 --model_name can_arizona_tea_norm". But the Segmentation fault poped out. Details are as below:

`(base) root@Friday:/home/chao/GitHub_repo/BundleTrack/scripts# sh run_nocs.sh 
K
591.013       0 322.525
      0 590.168 244.111
      0       0       1

Segmentation fault (core dumped)
`

The error message is a bit ambigious. Any idea on how to solve this?

Thanks in advance!

DCNSW, 2021-11-17:
@wenbowen123

wenbowen123, 2021-11-21:
@DCNSW @mkt1412 
Thanks for reporting this. Code has been updated. Please run `git pull -X theirs` and see if the problem goes away. If it still happens, please try to locate where it happens.

closed 2021-11-29 by wenbowen123