wenbowen123/BundleTrack #9: frame %s and %s findNN too few match, %s status marked as FAIL

Fahrenheit200, 2021-11-25:
I have tried to run the code on both datasets, but both can't produce the correct result. The error message is produced by SiftManager::findCorresbyNN. It seems that it can't find any match points between any two frames. The size of _matches[{frameA,frameB}] is always 0. And the pose data produced finally are all the same, same as the initial pose. Could you please help me figure out the problem？@wenbowen123

wenbowen123, 2021-11-29:
Can you check the output folder /tmp/BundleTrack?
There are some segmentation or keypoints visualization. Make sure they look good.

Fahrenheit200, 2021-11-29:
@wenbowen123 I call SiftManager::vizKeyPoints and get the keypoints.jpg. And there's no problem wtih the keypoints detection. It can actually detect 500 keypoints. I think there is something wrong wtih findCorresbyNN. It can't do the feature correspondense correctly.

wenbowen123, 2021-12-04:
In the config file, change the LOG to 2 and there will be more files logged in `/tmp/BundleTrack` for debugging. e.g. if you are running on NOCS, change this https://github.com/wenbowen123/BundleTrack/blob/48272977b3280da12c18cd10da03d08500540a4f/config_nocs.yml#L8

See if there is anything abnormal.

liuzheMET, 2021-12-16:
I found that the depth map was set to zero, when the program use cuda to preprocess it

wenbowen123, 2021-12-17:
@liuzheMET Can you upload (e.g. Google drive) your logged files so that I can take further look?

wenbowen123, 2021-12-28:
@liuzheMET @Fahrenheit200 
This seems related https://github.com/wenbowen123/BundleTrack/issues/15

Hanziscool, 2021-12-28:
@wenbowen123 i ran into the same issue again after #15

running 
```
python scripts/run_ycbineoat.py --data_dir /home/jrojas/YCBInEOAT/bleach0 --port 5555 --model_name 021_bleach_cleanser --model_dir /home/jrojas/YCBInEOAT/YCB_Video_Dataset/models/021_bleach_cleanser/textured_simple.obj
```

[image]
 my ycb config file
[image]
[image]
[image]
[image]

wenbowen123, 2021-12-28:
@Hanziscool let's discuss this in that issue.

kausiksivakumar, 2022-01-25:
I have the same issue with NOCS, I found nothing except poses saved in the debug directory.

kausiksivakumar, 2022-01-27:
[image]
This is one of the visualizations in NOCS. I am attaching the debug output here
https://drive.google.com/drive/folders/1qz2XRhPVkEL3-zns36scEWF3-RE3mBZN?usp=sharing

kausiksivakumar, 2022-02-01:
> I found that the depth map was set to zero, when the program use cuda to preprocess it

@liuzheMET Could you let me know where did you find that the depth map was set. Thanks

closed 2022-03-26 by wenbowen123

foww-0001, 2022-10-25:
> @wenbowen123 I call SiftManager::vizKeyPoints and get the keypoints.jpg. And there's no problem wtih the keypoints detection. It can actually detect 500 keypoints. I think there is something wrong wtih findCorresbyNN. It can't do the feature correspondense correctly.

I have the same problem, i visual the keypoint result, it can detect 500 keypoints, and when it send to pruneMatches function, matches_AB will be zero.