wenbowen123/BundleTrack #10: run the command "python lf-net-release/run_server.py"  erro"ModuleNotFoundError: No module named 'mso_resnet_detector'".why?

lv-lan, 2021-11-28:
```
Traceback (most recent call last):
  File "lf-net-release/run_server.py", line 113, in <module>
    ops = build_networks(config, photo_ph, is_training)
  File "lf-net-release/run_server.py", line 30, in build_networks
    DET = importlib.import_module(config.detector)
  File "/opt/conda/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'mso_resnet_detector'
```

Rolling998, 2021-11-29:
I meet the same problem, did you solve it?
[image]

wenbowen123, 2021-11-29:
If you are under bundletrack root directory, please try `cd lf-net-release && python run_server.py` 
Also see updated readme.

Rolling998, 2021-11-30:
It has been solved, thank you!

closed 2021-12-04 by wenbowen123