wenbowen123/BundleTrack #11: Opencv errors conduct while running on the YCBInEOAT dataset.

DCNSW, 2021-11-29:
[image]

Rolling998, 2021-11-30:
Also the same error. Sorry to bother @wenbowen123 
[image]

wenbowen123, 2021-12-04:
@Rolling998 @DCNSW do you have GPU on your computer? If so, which GPU is that?

DCNSW, 2021-12-05:
@wenbowen123  NVIDIA GeFore RTX 3090

wenbowen123, 2021-12-14:
@DCNSW @Rolling998 I was running on 2080 Ti. NVIDIA 3090 is a bit new. For 3090, you might need to build opencv from source with some additional settings for CUDA compilation inside docker container. I might provide an updated version for 3090 some time when I'm more free.

DCNSW, 2021-12-14:
Thanks, this issue should be closed.

closed 2021-12-14 by DCNSW

Rolling998, 2021-12-16:
OK, thank you very much

Hanziscool, 2021-12-21:
@DCNSW @Rolling998 have u guys got this running with building opencv from source? I am also having the same issue but with a newer NVIDIA A6000

wenbowen123, 2021-12-25:
@DCNSW @Rolling998  
Here is a newer solution
https://github.com/wenbowen123/BundleTrack/issues/15