wenbowen123/BundleTrack #12: SyntaxError: invalid syntax

Jie-fafa, 2021-12-08:
python version：3.8.8
sudo python scripts/run_nocs.py --nocs_dir /data/Jieie/NOCS --scene_id 1 --port 5555 --model_name can_arizona_tea_norm

 File "scripts/run_nocs.py", line 58
    cur_out_dir = f'/tmp/BundleTrack/nocs/{model_name}_{scene_id}'
                                                                 ^
SyntaxError: invalid syntax

wenbowen123, 2021-12-14:
Hi, what if you run without `sudo`?

Jie-fafa, 2021-12-14:
PermissionError: [Errno 13] Permission denied: '/tmp/config_model_can_arizona_tea_norm_scene1.yml'

wenbowen123, 2021-12-14:
It seems like you are not running inside the docker container. Can you take a screenshot of your terminal? It should be the same terminal where you ran `bash docker/run_container.sh` in previous steps.

Jie-fafa, 2021-12-14:
K
591.013       0 322.525
      0 590.168 244.111
      0       0       1

ob_in_cam0
  0.949412 -0.0346172  -0.312118   0.363097
  0.127492  -0.865821   0.483839 -0.0664708
 -0.286987  -0.499155  -0.817608   0.946327
         0          0          0          1

[pcl::OBJReader::readHeader] Could not find file '/data/Jieie/NOCS/obj_models/real_test/can_arizona_tea_norm.obj'.
[pcl::OBJReader::read] Problem reading header!
[pcl::OBJReader::readHeader] Could not find file '/data/Jieie/NOCS/obj_models/real_test/can_arizona_tea_norm.obj'.
[pcl::OBJReader::read] Problem reading header!

wenbowen123, 2021-12-17:
Can you make sure you have the file under `/data/Jieie/NOCS/obj_models/real_test/can_arizona_tea_norm.obj`?

Jie-fafa, 2021-12-23:
I cann't find this file.where can I download the file?

wenbowen123, 2021-12-24:
You need to first download the NOCS dataset. Please follow the dataset setup steps.

closed 2021-12-28 by wenbowen123