wenbowen123/BundleTrack #13: Unable to install bundletrack

alishan2040, 2021-12-17:
Hello,

Thanks for providing the implementation for Bundle Track, you work is amazing. While I was trying this project, I'm repeatedly facing the issue with opencv. I've already compiled opencv 4.5 from source but the issue persists. 
[image]

Can you please guide which version of opencv should work with bundle track. Also if you can include the requirements.txt file to the project, would be great.

Thanks,
Have a great day!,
Ali

wenbowen123, 2021-12-17:
Hi, have you tried using the docker container, where the dependencies should have been set up? For opencv version, I remember it's 3.3.X. But it's not recommended to setup the env yourself as there would be quite some overhead.

alishan2040, 2021-12-18:
Hi @wenbowen123, thanks for the reply.
I'm facing this issue when running this command **rm -rf build && mkdir build && cd build && cmake .. && make** as mentioned in the instructions. Should I skip this command,  and follow the rest?. I'm beginner to docker so have not clear idea about its usage.

Thanks

tengxinyu151, 2021-12-22:
> Hello,
>
> Thanks for providing the implementation for Bundle Track, you work is amazing. While I was trying this project, I'm repeatedly facing the issue with opencv. I've already compiled opencv 4.5 [...]

Hello! I am also trying this excellent project. This  is due to the copyright problem. For opencv versions above 4.3, just change it to cv::SIFT::create().

wenbowen123, 2021-12-24:
@alishan2040 sift used to be in opencv-contrib. What @tengxinyu151 mentioned is correct. 
You can also safely comment that line since it's not used.

closed 2021-12-28 by wenbowen123