wenbowen123/BundleTrack #16: Segmentation fault - when running predictions on NOCS

kausiksivakumar, 2022-01-04:
Running command 
`python scripts/run_nocs.py --nocs_dir [PATH_TO_NOCS] --scene_id 1 --port 5555 --model_name can_arizona_tea_norm`
results in the following output 
[image]

wenbowen123, 2022-01-05:
Did you download the NOCSdataset as the first step in "Run predictions on NOCS" in readme?

kausiksivakumar, 2022-01-05:
Yes I did, they only contained these files 
[image]
the file `real_test/can_arizona_tea_norm.obj` is not present

wenbowen123, 2022-01-07:
Thanks for reporting this! Please check the updated readme and let me know if there is any problem.

kausiksivakumar, 2022-01-07:
Yes, I have updated it. I was able to find 'can_arizona_tea_norm.obj'. But the .mtl files were missing in the same. I am attaching the screenshot here
[image]
Not sure how to go about this, thanks for helping

wenbowen123, 2022-01-08:
Good to see it works. The .mtl file should not affect running. Feel free to ignore it.

closed 2022-01-08 by wenbowen123

kausiksivakumar, 2022-01-11:
Thanks, I tried going to the location where the results would be present `/tmp/BundleTrack`. After running the script `run_nocs.py` as specified in the instructions. I found that the file itself is empty. Is this okay? 
[image]
I thought that we would be getting the 6D poses of the objects whose masks were provided in the results. Did I get it wrong? Could someone help with this, thanks