wenbowen123/BundleTrack #19: No files present in /tmp/Bundletrack

kausiksivakumar, 2022-01-11:
With reference to #16, I tried going to the location where the results would be present `/tmp/BundleTrack`. After running the script `run_nocs.py` as specified in the instructions. I found that the file itself is empty. Is this okay? 
[image]
I thought that we would be getting the 6D poses of the objects whose masks were provided in the results. Did I get it wrong? Could someone help with this, thanks

_Originally posted by @kausiksivakumar in https://github.com/wenbowen123/BundleTrack/issues/16#issuecomment-1010355377_

wenbowen123, 2022-01-13:
Hi, by default these visualization results are not enabled to be saved as they make the program run slower. In order to save them, including the segmentated image, you need to do this: 
https://github.com/wenbowen123/BundleTrack/issues/15#issuecomment-1004555934

kausiksivakumar, 2022-01-14:
Oh I understand, could you also clarify what should be expected in the results as mentioned in the instructions. Because in the case of NOCS dataset predictions, I get (https://github.com/wenbowen123/BundleTrack/issues/16#issuecomment-1007743114) #16, to just confirm this feedback is alright right?

closed 2022-01-19 by kausiksivakumar