wenbowen123/BundleTrack #22: Can't find NOCS visualizations and path to the runnning_outputs

kausiksivakumar, 2022-01-25:
In the instructions mentioned for running evaluations on NOCS, what is the directory of [PATH TO THE RUNNING OUTPUTS]? 
Also while running predictions on **NOCS**, I followed #15 's instructions and updated **Bundler.cpp** accordingly. Even after doing that, I could not find the the visualizations of NOCS stored in the results directory. I could only find a directory called poses that contains all the homogeneous transforms in a txt file. The directory **color_viz** does not contain any file. Could you help me with it. Thanks

I did three things, 
1) set log = 3 (in config_nocs.yml)
2) Add **_fm->vizKeyPoints(frame);** in Bundler.cpp
3) I did not quite understand what you were going for here https://github.com/wenbowen123/BundleTrack/issues/15#issuecomment-1004555934 . Could you clarify how to save the image?
This is my output that I received at terminal while running predictions on NOCS

```

New frame 0280
zmq start waiting for reply
zmq got reply
finding corres between 0280(id=1) and 0000(id=0)
frame 0280 and 0000 findNN too few match, 0280 status marked as FAILcolor file: ./NOCS/real_test/scene_1//../../real_test/scene_1/0281_color.png

```
The above message is for all the frames
Could you help me with this. Thanks

kausiksivakumar, 2022-01-25:
Does it have something to do with the commented part of the code 
[image]

kausiksivakumar, 2022-01-25:
Also me changing debug_dir on **config_nocs.yml** doesn't seem to change the directory in which the results are stored

kausiksivakumar, 2022-01-25:
Also I tried changing **log  = 3** in run_nocs.py manually
[image]
Now, when I run again, I get this as feedback
[image]

kausiksivakumar, 2022-01-25:
But me changing log = 3 in **config_nocs.yml** is not changing anything in the feedback. 
[image]
Could you please help me with this @wenbowen123 . I sincerely appreciate your time and effort. Thank you

wenbowen123, 2022-01-30:
Hi, sorry for the late reply.

PATH TO THE RUNNING OUTPUTS: this refers to the debug_dir [in config_nocs.yml](https://github.com/wenbowen123/BundleTrack/blob/476d56f38527de1029306df6d9699151907df568/config_nocs.yml#L6)

> But me changing log = 3 in **config_nocs.yml** is not changing anything in the feedback. [image] Could you please help me with this @wenbowen123 . I sincerely appreciate your time and effort. Thank you

What command did you run after it? Can you elaborate your complete steps. In particular, I 

> Also I tried changing **log = 3** in run_nocs.py manually [image] Now, when I run again, I get this as feedback [image]

This should be the right way to do, but I'm not sure why you run into that error. Can you use some printing function to get some clues where it happens?

wenbowen123, 2022-01-30:
Can you show me the file structures under `debug_dir: /tmp/BundleTrack/ ` if you didn't change the debug_dir.

kausiksivakumar, 2022-02-01:
I did change the current output directory in **run_nocs.py** ans then changed the logs to 3

kausiksivakumar, 2022-02-01:
When I change log to 3 in 
[image]
**config_nocs.py** i am able to get keypoints visualized but it is unable to find matches between images. 
[image]
I'm basically getting this error
Debug_dir: https://drive.google.com/drive/folders/1qz2XRhPVkEL3-zns36scEWF3-RE3mBZN?usp=sharing

wenbowen123, 2022-02-02:
Yes, the keypoints detection look good. It should be the matching problem.

Can you change this line to =1 and re-build and run?
https://github.com/wenbowen123/BundleTrack/blob/476d56f38527de1029306df6d9699151907df568/CMakeLists.txt#L19

kausiksivakumar, 2022-02-03:
Oh, it is already = 1

kausiksivakumar, 2022-02-03:
I get this when I turn it to 0 
[image]

wenbowen123, 2022-02-05:
It is strange that you got frame 389 right after 0. Are the RGB files in your image folder complete and ordered?

kausiksivakumar, 2022-02-05:
Yes they are. This matching error occurs at all the frames from 1 to 389. 

[image]
This continues till frame 389

wenbowen123, 2022-02-06:
At https://github.com/wenbowen123/BundleTrack/blob/476d56f38527de1029306df6d9699151907df568/src/FeatureManager.cpp#L247

can you replace with this code then re-build and run and copy the output here?
```
void SiftManager::findCorresbyNN(std::shared_ptr<Frame> frameA, std::shared_ptr<Frame> frameB)
{
  assert(frameA->_id > frameB->_id);
  if (frameA->_keypts.size()==0 || frameB->_keypts.size()==0) return;

  const float max_dist_no_neighbor = (*yml)["feature_corres"]["max_dist_no_neighbor"].as<float>();
  const float cos_max_normal_no_neighbor = std::cos((*yml)["feature_corres"]["max_normal_no_neighbor"].as<float>()/180.0*M_PI);
  const float max_dist_neighbor = (*yml)["feature_corres"]["max_dist_neighbor"].as<float>();
  const float cos_max_normal_neighbor = std::cos((*yml)["feature_corres"]["max_normal_neighbor"].as<float>()/180.0*M_PI);

  const int H = frameA->_H;
  const int W = frameA->_W;

  bool is_neighbor = std::abs(frameA->_id-frameB->_id)==1;

  std::vector<cv::DMatch> matches_AB, matches_BA;
  std::vector< std::vector<cv::DMatch> > knn_matchesAB, knn_matchesBA;
  const int k_near = 5;

#if NO_OPENCV_CUDA
  cv::Ptr<cv::DescriptorMatcher> matcher = cv::DescriptorMatcher::create(cv::DescriptorMatcher::BRUTEFORCE);
  matcher->knnMatch( frameA->_feat_des, frameB->_feat_des, knn_matchesAB, k_near);
  matcher->knnMatch( frameB->_feat_des, frameA->_feat_des, knn_matchesBA, k_near);
#else
  cv::Ptr<cv::cuda::DescriptorMatcher> matcher = cv::cuda::DescriptorMatcher::createBFMatcher(cv::NORM_L2);
  matcher->knnMatch( frameA->_feat_des_gpu, frameB->_feat_des_gpu, knn_matchesAB, k_near);
  matcher->knnMatch( frameB->_feat_des_gpu, frameA->_feat_des_gpu, knn_matchesBA, k_near);
#endif

  printf("knn_matchesAB: %d, knn_matchesBA: %d\n",knn_matchesAB.size(),knn_matchesBA.size());
  pruneMatches(frameA,frameB,knn_matchesAB,matches_AB);

  pruneMatches(frameB,frameA,knn_matchesBA,matches_BA);
  printf("111 knn_matchesAB: %d, knn_matchesBA: %d\n",knn_matchesAB.size(),knn_matchesBA.size());

  collectMutualMatches(frameA,frameB,matches_AB,matches_BA);

  if (_matches[{frameA,frameB}].size()<5 && is_neighbor)
  {
    frameA->_status = Frame::FAIL;
    printf("frame %s and %s findNN too few match, %s status marked as FAIL", frameA->_id_str.c_str(), frameB->_id_str.c_str(), frameA->_id_str.c_str());
  }

}
```

kausiksivakumar, 2022-02-06:
Yes, this is what I received for all the frames 
[image]

kausiksivakumar, 2022-02-08:
Is there something going wrong with the way I have installed packages?

wenbowen123, 2022-02-09:
This is strange. I tested a while ago and didn't have this error. Can you update with this and repeat the steps? Also make sure you didn't change the code beyond the configuration files.
```
void SiftManager::findCorresbyNN(std::shared_ptr<Frame> frameA, std::shared_ptr<Frame> frameB)
{
  assert(frameA->_id > frameB->_id);
  if (frameA->_keypts.size()==0 || frameB->_keypts.size()==0) return;

  const float max_dist_no_neighbor = (*yml)["feature_corres"]["max_dist_no_neighbor"].as<float>();
  const float cos_max_normal_no_neighbor = std::cos((*yml)["feature_corres"]["max_normal_no_neighbor"].as<float>()/180.0*M_PI);
  const float max_dist_neighbor = (*yml)["feature_corres"]["max_dist_neighbor"].as<float>();
  const float cos_max_normal_neighbor = std::cos((*yml)["feature_corres"]["max_normal_neighbor"].as<float>()/180.0*M_PI);

  const int H = frameA->_H;
  const int W = frameA->_W;

  bool is_neighbor = std::abs(frameA->_id-frameB->_id)==1;

  std::vector<cv::DMatch> matches_AB, matches_BA;
  std::vector< std::vector<cv::DMatch> > knn_matchesAB, knn_matchesBA;
  const int k_near = 5;

#if NO_OPENCV_CUDA
  cv::Ptr<cv::DescriptorMatcher> matcher = cv::DescriptorMatcher::create(cv::DescriptorMatcher::BRUTEFORCE);
  matcher->knnMatch( frameA->_feat_des, frameB->_feat_des, knn_matchesAB, k_near);
  matcher->knnMatch( frameB->_feat_des, frameA->_feat_des, knn_matchesBA, k_near);
#else
  cv::Ptr<cv::cuda::DescriptorMatcher> matcher = cv::cuda::DescriptorMatcher::createBFMatcher(cv::NORM_L2);
  matcher->knnMatch( frameA->_feat_des_gpu, frameB->_feat_des_gpu, knn_matchesAB, k_near);
  matcher->knnMatch( frameB->_feat_des_gpu, frameA->_feat_des_gpu, knn_matchesBA, k_near);
#endif

  printf("knn_matchesAB: %d, knn_matchesBA: %d\n",knn_matchesAB.size(),knn_matchesBA.size());
  pruneMatches(frameA,frameB,knn_matchesAB,matches_AB);

  pruneMatches(frameB,frameA,knn_matchesBA,matches_BA);
  printf("111 knn_matchesAB: %d, knn_matchesBA: %d\n",knn_matchesAB.size(),knn_matchesBA.size());

  collectMutualMatches(frameA,frameB,matches_AB,matches_BA);

  if (_matches[{frameA,frameB}].size()<5 && is_neighbor)
  {
    frameA->_status = Frame::FAIL;
    printf("frame %s and %s findNN too few match: %d, %s status marked as FAIL\n", frameA->_id_str.c_str(), frameB->_id_str.c_str(), _matches[{frameA,frameB}].size(), frameA->_id_str.c_str());
  }

}
```

kausiksivakumar, 2022-02-09:
Hmm, I still get the same error
[image]
But that said, I did change a few things. For instance, I changed data, mask and debug directory in the yml file
[image]
I found that the logs and the debug directory were getting overwritten so I had to change it in **run_nocs.py** file
[image]
And I added 
[image]
in **Bundler.cpp**

kausiksivakumar, 2022-02-09:
Should I try reinstalling the process from scratch again?

wenbowen123, 2022-02-11:
> I found that the logs and the debug directory were getting overwritten so I had to change it in run_nocs.py file

Hi, can you clarify how are these over-written (or over-written by what)? It should change the output folder of BundleTrack so the results will dumped there. 

Can you replace collectMutualMatches with below and repeat the previous steps? We might be close to find the problem.
```
void SiftManager::collectMutualMatches(std::shared_ptr<Frame> frameA, std::shared_ptr<Frame> frameB, const std::vector<cv::DMatch> &matches_AB, const std::vector<cv::DMatch> &matches_BA)
{
  auto &matches = _matches[{frameA, frameB}];

  for (int i=0;i<matches_AB.size();i++)
  {
    int Aid = matches_AB[i].queryIdx;
    int Bid = matches_AB[i].trainIdx;
    float uA = frameA->_keypts[Aid].pt.x;
    float vA = frameA->_keypts[Aid].pt.y;
    float uB = frameB->_keypts[Bid].pt.x;
    float vB = frameB->_keypts[Bid].pt.y;
    const auto &ptA = frameA->_cloud->at(std::round(uA), std::round(vA));
    const auto &ptB = frameB->_cloud->at(std::round(uB), std::round(vB));
    Correspondence corr(uA,vA,uB,vB, ptA, ptB, true);
    matches.push_back(corr);
  }
  printf("#matches=%d\n",matches.size());
  for (int i=0;i<matches_BA.size();i++)
  {
    int Aid = matches_BA[i].trainIdx;
    int Bid = matches_BA[i].queryIdx;
    float uA = frameA->_keypts[Aid].pt.x;
    float vA = frameA->_keypts[Aid].pt.y;
    float uB = frameB->_keypts[Bid].pt.x;
    float vB = frameB->_keypts[Bid].pt.y;
    const auto &ptA = frameA->_cloud->at(std::round(uA), std::round(vA));
    const auto &ptB = frameB->_cloud->at(std::round(uB), std::round(vB));
    Correspondence corr(uA,vA,uB,vB, ptA, ptB, true);
    matches.push_back(corr);
  }
  printf("111 #matches=%d\n",matches.size());
}
```

kausiksivakumar, 2022-02-15:
> 

The part that overwrites the waml file is in run_nocs.py (in run_one_scene_one_model function)
[image]

kausiksivakumar, 2022-02-15:
I changed CollectMutualMatches with the script given 
[image]
I got this for all frame combination

wenbowen123, 2022-02-15:
https://github.com/wenbowen123/BundleTrack/issues/9#issuecomment-995455933
this might be the problem. Are you running with a GPU and does the cuda program compile successfully?

kausiksivakumar, 2022-02-15:
https://github.com/wenbowen123/BundleTrack/issues/22#issuecomment-1028387412 
This only works when I set this to 1

wenbowen123, 2022-02-15:
Your laptop has `NVIDIA GeForce RTX 3050 Ti` which is relatively new. I remember some users have met similar issues using the newer GPUs. This repo has been tested on 2080 Ti. Other older GPUs may also work. I don't have newer GPUs to develop and test on my side, so there is not plan of supporting the new GPUs in near future. Another workaround will be rewriting all the cuda code by C++.

wenbowen123, 2022-02-23:
We used driver 460 and CUDA 11.2

kausiksivakumar, 2022-02-24:
Yes, it works. I believe as you said, the problem is with the newer GPUs. Also, the path to download the weights returns a 404 now. Thought I should bring it to your notice

closed 2022-02-24 by wenbowen123