wenbowen123/BundleTrack #25: Structure of files for YCBINEOAT data directory

kausiksivakumar, 2022-02-28:
Hi Bowen, 
Could you let me know how the files should be stored i the YCBINEOAT directory for prediction. Like how you showed it for NOCS 
[image]
Also could you provide an example of what the model_name and model_directory should be changed to in the same. 
Thanks much!

wenbowen123, 2022-02-28:
Hi, readme is just updated. Please check and let me know if there is any question.

kausiksivakumar, 2022-03-01:
Thanks much!. I also have a few doubts in regards to running YCB predictions. 

`python scripts/run_ycbineoat.py --data_dir ./Data/YCBINEOAT/bleach0 --port 5555 --model_name 021_bleach_cleanser
/home/kausik/workspace/BundleTrack/scripts/../build/bundle_track_ycbineoat /tmp/config_YCBINEOAT.yml
cam K=
319.582       0 320.215
      0 417.119 244.349
      0       0       1
ob_in_cam0
-0.698192 0.0511303  0.714199 -0.393181
 0.572012  0.639666    0.5133  0.274228
-0.430625  0.766652 -0.475893  0.836339
        0         0         0         1

[pcl::OBJReader::readHeader] Could not find file '/media/bowen/e25c9489-2f57-42dd-b076-021c59369fec/DATASET/YCB_Video_Dataset/CADmodels/021_bleach_cleanser/textured.obj'.
[pcl::OBJReader::read] Problem reading header!
[pcl::OBJReader::readHeader] Could not find file '/media/bowen/e25c9489-2f57-42dd-b076-021c59369fec/DATASET/YCB_Video_Dataset/CADmodels/021_bleach_cleanser/textured.obj'.
[pcl::OBJReader::read] Problem reading header!
data has 663 images
Connected to port 5555
color file: ./Data/YCBINEOAT/bleach0/rgb/1578966042136484471.png

New frame 1578966042136484471
mask file open failed: ./Data/YCBINEOAT/bleach0/masks/1578966042136484471.png`

I got this when running the code given. There is no directory named bleach0/masks/ in YCB data directory. There is a directory called bleach0/gt_masks though, is this what should be passed in the code?

wenbowen123, 2022-03-01:
gt_mask is not. I'll upload the correct masks recently, probably this week.

wenbowen123, 2022-03-01:
Just realize it's already there https://github.com/wenbowen123/BundleTrack#data

closed 2022-03-01 by wenbowen123