wenbowen123/BundleTrack #29: Running BundleTrack online

kausiksivakumar, 2022-04-09:
Hey Bowen, in case I want to run Bundletrack online, how should I go for it. Would you have any thoughts

wenbowen123, 2022-04-11:
Hi, currently the video segmentation network is implemented to run beforehand over the video to get the mask for each frame when given the first frame's mask as input. To run online, you can either make it run online or consider alternatives, e.g. for robotic manipulation case, one can also use simple 3D box cropping to get the rough mask since the forward kinematics is known.

kausiksivakumar, 2022-04-11:
I understand. Could you let me know if I am wrong, I had presumed that the mask of the object initially was enough for BundleTrack to be able to track the object. So the pipeline would be, feed images with initial mask to the `transducive-vos-pytorch` to generate mask for the entire rgb image data and this is fed as input to the BundleTrack code. 
Also, what is gt_masks in the YCB dataset?

wenbowen123, 2022-04-12:
Your understanding is correct.

`gt_mask` is not used and can be ignored.

closed 2022-04-12 by wenbowen123