wenbowen123/BundleTrack #30: Running prediction on own dataset

kausiksivakumar, 2022-04-10:
Hey Bowen, 
When I try to run predictions on my own data, I get a segmentation fault core dumped error 
[image]
Do you happen to know if anyone else faced something similar. I tried making the directory as similar to bleack0 (which worked really well though, so I doubt there is something going wrong in installation)

wenbowen123, 2022-04-11:
If you are able to run successfully with my example bleach0, then there should be something wrong with your data. I'd suggest you inspect that, especially make sure your masks are correct.  

There are similar issues as well as details about how to debug/visualize, such as https://github.com/wenbowen123/BundleTrack/issues/22

kausiksivakumar, 2022-04-11:
I see I think I got it wrong while getting the initial mask. It runs now, but whenever I run `transducive-vos-pytorch`, I get the mask for a few frames and for the rest, I get the CUDA ran out of memory error. The scene is not that complicated, is there anything that you think could be done here. Also thanks so much for helping, I sincerely appreciate it

wenbowen123, 2022-04-12:
Indeed, the segmentation network take some GPU memory. We had 12G. If yours is lower than that, it may not be able to afford the network. In that case, consider resize your images smaller, but worse performance is expected. Hopefully your scene is simple enough.

closed 2022-04-19 by wenbowen123