wenbowen123/BundleTrack #32: Segmentation fault

Spj-Zhao, 2022-04-27:
(base) root@hung-Alienware-Aurora-R9:/home/hung/Desktop/BundleTrack# python scripts/run_nocs.py --nocs_dir /home/hung/Desktop/BundleTrack/NOCS --scene_id 1 --port 5555 --model_name can_arizona_tea_norm
/home/hung/Desktop/BundleTrack/scripts/../build/bundle_track_nocs /tmp/config_model_can_arizona_tea_norm_scene1.yml
K
591.013       0 322.525
      0 590.168 244.111
      0       0       1

Reading directory failed: /home/hung/Desktop/BundleTrack/NOCS/gts/real_test_text/scene_1/model_can_arizona_tea_norm/
Segmentation fault (core dumped)

could you please upload the _NOCS/gts/real_test_text/scene_1/model_can_arizona_tea_norm/_

wenbowen123, 2022-04-27:
See if this https://github.com/wenbowen123/BundleTrack/issues/31#issuecomment-1111127391 works

Spj-Zhao, 2022-04-28:
> See if this [#31 (comment)](https://github.com/wenbowen123/BundleTrack/issues/31#issuecomment-1111127391) works

Thanks a lot,it's solved.

closed 2022-05-03 by wenbowen123