wenbowen123/BundleTrack #34: Segmentation on YCBInEOAT

WilliamKRobert, 2022-05-05:
Hi, I noticed that the ground truth segmentation in YCBInEOAT dataset seems not good. The segmentation tracking by VOS is even worse. For example, if we look at "bleach0", the object is partially segmented with only a few points. How does BundleTrack handle this scenario?

wenbowen123, 2022-05-05:
Hi, yes the scene is quite challenging. If you look at the masks (not gt_masks which is not used), the segmentation is very noisy. However, BundleTrack turns out to be robust to the noisy segmentations like that.

WilliamKRobert, 2022-05-06:
Thanks for the clarification. Is the segmentation in "ycbineoat_ours_results.tar.gz/color_raw" produced by Transductive VOS network?

WilliamKRobert, 2022-05-06:
I just read your documentation again and found, on YCBInEOAT, segmentation is calculated from forward kinetics, instead of VOS network.

wenbowen123, 2022-05-06:
you understanding is correct. This is also mentioned in the paper. The segmentation module is flexible and can use whatever suitable.

closed 2022-05-06 by wenbowen123