wenbowen123/BundleTrack #36: Core dump problem

YYYYYHC, 2022-05-10:
Hi, after setting up the environment, I am caught with a segmentation fault.
The problem happens when the texture.obj is loaded into a PolygonMesh object by pcl::io::loadOBJFile (Dataloader.cpp line318). The terminal output is _Segmentation fault (core dumped)_. I am wondering if the textured.obj is too big for cpus to load.
I notice that you are using texture_simple.obj rather than the texture.obj, could you provide me the .obj files you have tested ( e.g. 021_bleach_cleanser) ?

wenbowen123, 2022-05-10:
Hi, the texture_simple.obj can be obtained from YCBVideo dataset
https://drive.google.com/file/d/1gmcDD-5bkJfcMKLZb3zGgH_HUFbulQWu/view

YYYYYHC, 2022-05-11:
> Hi, the texture_simple.obj can be obtained from YCBVideo dataset
> https://drive.google.com/file/d/1gmcDD-5bkJfcMKLZb3zGgH_HUFbulQWu/view

Very helpful, thanks a lot!

closed 2022-05-11 by YYYYYHC