wenbowen123/BundleTrack #38: About /pose/*.txt

YYYYYHC, 2022-05-13:
Hi Bowen, it's me again ^ ^.
I have successfully run the pipeline of bundletrack on my own data. I input the initial pose (world coordinate sys) and get the output poses sequence (/pose/000-999.txt), the content of the txt file are 4*4 transformation matrices.
Here is one thing I want to know: which coordinate system are those transformation matrices in? My ground truth labels are set under the world coordinate sys, so if these transformations are in another system, could you tell me how to transfer them into world coordinate system? That would be very helpful to me!
Thanks again for your great work and patience!

wenbowen123, 2022-05-13:
You can use the following formula
`pred_new = (pred_pose @ np.linalg.inv(init_pose)) @ init_pose_new`
where `pred_pose` is the raw output you have for some time stamp, e.g. `/pose/123.txt`, `init_pose` in the first frame's output `/pose/0.txt`,  `init_pose_new` is the transformation from your desired coordinate system to the camera's coordinate at initial time stamp.

YYYYYHC, 2022-05-14:
> pred_new = (pred_pose @ np.linalg.inv(init_pose)) @ init_pose_new
I'm sorry, but the formula doesn't make sense to me. 
First, I find in the code that inv(init_pose) seems to be my input initial pose. I am wondering **what is the exact meaning of pred_pose in /pose/~.txt**. Is it the transformation from the initial frame to the current frame in the camera coordinate system?
Second, I can't understand why the init_pose_new (extrinsic camera matrix in our settings) is multiplied at right. 
Could you explain the meaning of these transformation matrices further? I have been thinking it all day and just can't get it.

wenbowen123, 2022-05-14:
BundleTrack predicts relative motions from the initial time stamp. By default, bundletrack doesn't take your input, so the `init_pose` should be bundletrack's internal initial pose of the unknown object, where the coordinate frame is set to the same as camera. That's why there is `init_pose_new` that allows you to provide. It's not the extrinsic matrix, but the new definition of your object coordinate frame (For unknown object, there is no unique way to set the coordinate frame and depending on your own use case, you can e.g. set the point cloud origin as the coordinate system's origin). The maths behind this is shown in the paper problem statement section. The transformation indeed is a bit hard to understand. I'd suggest you first try this formula and see if it's really what you want. Once you have the transformation from your defined object coordinate system to camera and if you have calibrated the camera to the world, you can easily get the object pose in the world.

Also it'd be more clear if you can elaborate what's your world coordinate frame and what's your use case?

YYYYYHC, 2022-05-14:
Since I am testing BundleTrack in a simulation environment, there is  an absolute world coordinate system and all the pose (both the initial pose and ground truth pose for evaluation) are set under the fixed world coordinate system. Our camera are moving in the world coordinate, together with the target tracking object. We also have the extrinsic matrix of every frame.
Under our case, what should my input **initial_pose** be and how could I calculate the **init_pose_new**?
Thank you again for the response!😣

wenbowen123, 2022-05-14:
Like above mentioned, `init_pose` in the first frame's output from BundleTrack: /pose/0.txt. It's always that one, regardless of yours.
In your case, init_pose_new should be the object's pose in the camera's coordinate system at the beginning. BTW, is your target object also moving in the world?

YYYYYHC, 2022-05-15:
Thanks for the response!
Yes, both the camera and the target object are moving in the world.

YYYYYHC, 2022-05-16:
I have another question here:
If I apply the formula
`pred_new = (pred_pose @ np.linalg.inv(init_pose)) @ init_pose_new`
to the initial frame (frame0), both the `pred_pose` and `init_pose` are the pose in /pose/0.txt.
So the `pred_new`  equals to the `init_pose_new` at first frame, which is

>  the object's pose in the camera's coordinate system at the beginning

as you mentioned above.
However, what I really want is the object's pose in the world coordinate system, but at first frame it seems that I have no way getting it with that formula.
I am wondering what's going wrong here, should I just skip the first frame and apply the formula to the rest and get poses in world coordinate system? That seems to be magical math somehow.....

YYYYYHC, 2022-05-16:
I can get good results with this formula:
```
pred_new = (pred_pose @ np.linalg.inv(init_pose)) @ init_pose_new
pred_new_world = np.linalg.inv(extrinsic) @ pred_new
```
They work quiet well in the test, do they make sense mathmatically?

wenbowen123, 2022-05-16:
The first frame shouldn't be estimated. That's up to your own defined coordinate system of the object `init_pose_new`. BundleTrack is a tracker, so the first frame is skipped, which is initialized from yours.

Your formula looks reasonable, if your `extrinsic` is the transformation from world to the camera.

YYYYYHC, 2022-05-16:
Yes, it works.Thank you so much for your patience!

closed 2022-05-16 by YYYYYHC

ebianchi, 2024-02-15:
Contributing some clarity here since this is a git issue I've returned to a couple times and wanted to sanity check the correctness each time:

> I can get good results with this formula:
> 
> ```
> pred_new = (pred_pose @ np.linalg.inv(init_pose)) @ init_pose_new
> pred_new_world = np.linalg.inv(extrinsic) @ pred_new
> ```
> 
> They work quiet well in the test, do they make sense mathmatically?

TLDR, yes.  :)

The working expression involves the below transformations, using 4-by-4 transformation matrix form:

- `camera_T_Bn` (`pred_pose`):  BundleTrack's reported pose of the BundleTrack origin at the nth time stamp, in camera frame.
- `camera_T_B1` (`init_pose`):  BundleTrack's reported pose of the BundleTrack at the first time stamp, in camera frame.
- `camera_T_D0` (`init_pose_new`):  The (known/manually provided) pose of the desired body origin at the 0th time stamp, in camera frame.
- `camera_T_Dn` (`pred_new`):  The pose of the desired body origin at the first time stamp, in camera frame.
- `world_T_Dn` (`pred_new_world`):  The pose of the desired body origin at the nth time stamp, in world frame.
- `world_T_camera`:  Required camera extrinsics.

### Step 1:  Tracking desired body origin in camera frame.
The first desired result is `camera_T_Dn`, which is obtained using the following identity:
```
Bn_T_Dn = B0_T_D0    # <-- due to rigid body constraints, relative origin offset fixed.
```

Thus `camera_T_Dn` is obtained via:
```
camera_T_Dn = camera_T_Bn * Bn_T_Dn
            = camera_T_Bn * B0_T_D0
            = camera_T_Bn * B0_T_camera * camera_T_D0
            = camera_T_Bn * inv(camera_T_B0) * camera_T_D0

Or pred_new =    pred     *  inv(init_pose)  * init_pose_new
```

### Step 2:  Tracking desired body origin in world frame.
Now that we have `camera_T_Dn`, the resulting conversion is straightforward.
```
world_T_Dn = world_T_camera * camera_T_Dn
```
Note that the above is only valid for a fixed camera.  However if the camera moves, then `world_T_camera0`, `world_T_camera1`, ... `world_T_cameran` is needed for all timesteps.