wenbowen123/BundleTrack #39: Setup：rm -rf build && mkdir build && cd build && cmake .. && make

Jie-fafa, 2022-05-14:
[  4%] Building NVCC (Device) object CMakeFiles/cuda_compile_1.dir/src/cuda/cuda_compile_1_generated_cuda_ransac.cu.o
nvcc fatal   : A single input file is required for a non-link phase when an outputfile is specified
CMake Error at cuda_compile_1_generated_cuda_ransac.cu.o.Release.cmake:220 (message):
  Error generating
  /data/Jieie/BundleTrack/build/CMakeFiles/cuda_compile_1.dir/src/cuda/./cuda_compile_1_generated_cuda_ransac.cu.o

CMakeFiles/BundleTrack.dir/build.make:125: recipe for target 'CMakeFiles/cuda_compile_1.dir/src/cuda/cuda_compile_1_generated_cuda_ransac.cu.o' failed
make[2]: *** [CMakeFiles/cuda_compile_1.dir/src/cuda/cuda_compile_1_generated_cuda_ransac.cu.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/BundleTrack.dir/all' failed
make[1]: *** [CMakeFiles/BundleTrack.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2

wenbowen123, 2022-05-14:
Updated. Can you re-clone this repo and try again?

Jie-fafa, 2022-05-17:
Thank you for your reply. I tried the old version of the repo.

lovesuiyue, 2022-05-22:
(base) root@zyb-pc:/home/bowen/BundleTrack# rm -rf build && mkdir build && cd build && cmake .. && make
CMake Error: The source directory "/home/bowen/BundleTrack" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

closed 2022-05-27 by wenbowen123