wenbowen123/BundleTrack #44: Run predictions on own RGBD data

Jie-fafa, 2022-05-19:
python transductive-vos.pytorch/run_video.py --img_dir /data/Jieie/BundleTrackData/YCBInEOAT/0311_9/depth --init_mask_file /data/Jieie/BundleTrackData/YCBInEOAT/0311_9/one_mask.png --mask_save_dir /data/Jieie/BundleTrackData

RuntimeError: CUDA error: no kernel image is available for execution on the device
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.

wenbowen123, 2022-05-21:
Do you have a GPU?

Jie-fafa, 2022-05-21:
Yes.But it's 3090

wenbowen123, 2022-05-27:
As discussed in some older issues. The repo currently doesn't support newer GPUs yet, such as 3090. Please try on 2080 Ti or older. We may add support to the new GPUs in the future.

closed 2022-06-08 by wenbowen123