wenbowen123/BundleTrack #45: 编译问题

lovesuiyue, 2022-05-22:
您好 ，编译是在docker环境编译还是在普通环境下编译，因为我在docker 环境下编译缺少cmakelist.txt文件。

wenbowen123, 2022-05-22:
Did you edit `docker/run_container.sh` with your paths?

lovesuiyue, 2022-05-22:
编译之后进入的是bash环境吗？

lovesuiyue, 2022-05-22:
跑您的catcrasp 的docker环境进入的是catcrasp  ，但是跑这个demo进入的是bash 环境

wenbowen123, 2022-05-22:
You should compile and run inside docker. Edit docker/run_container.sh with your paths so that you can find CMakelists.txt inside docker.

lovesuiyue, 2022-05-23:
wen 博士，Bundletrack文件夹里面有两个docker文件夹，我应该在哪个里面运行bash run_container.sh呢？

wenbowen123, 2022-05-23:
The one under root dir

lovesuiyue, 2022-05-23:
sorry,我运行之后还是找不到cmakelist.txt文件, 这是在bundletrack目录下运行run_container.sh出现的警告

WARNING: Your kernel does not support swap limit capabilities or the cgroup is not mounted. Memory limited without swap.

wenbowen123, 2022-05-27:
Did you edit docker/run_container.sh with your paths?

closed 2022-06-08 by wenbowen123