wenbowen123/BundleTrack #46: No visulization output on YCB and custom data

zituka, 2022-05-27:
Hi wenbowen :
    thx for your great job on bundletrack!
    I have some problem when running prediction on YCB data. There is no output on my /tmp/BundleTrack file but I get poses output txt.

[image]

and when I try to predict on my own rgbd data, it shows 

[image]

I donot understand why it need a obj file or annotation file?

wenbowen123, 2022-05-27:
The annotation file here refers to an initial pose. You can create a dummy txt file with identity 4x4 matrix, if your desired initial pose is set the same as camera. Or you can take the mean of the initial masked point cloud and set to the translation part in the initial pose matrix to set object center as the origin.

zituka, 2022-05-28:
Thanks for your reply!
Now I want to do a visualization on my own RGBD dataset, so if I add the pose annotation file it could be done right?
AND, There is nothing in the 'color_viz' file when I want to visualization on YCB dataset, (the first and second screen shot), how can I see the visualization output on YCB dataset?

closed 2022-07-01 by wenbowen123

PolarisXQ, 2023-09-21:
> Hi, if you provided the true mesh of the target object, as well as the correct initial pose, then you can visualize the point cloud overlay. You can try the example YCBInEOAT data to get an idea.
> 
> Yes, by default, the bounding box is not plotted.

Would you please be so kind to explain how to draw a bounding box or obtain bounding box results?

wenbowen123, 2023-09-21:
@PolarisXQ you can refer to https://github.com/NVlabs/BundleSDF/issues/91