wenbowen123/BundleTrack #47: Could not find *.mtl files in NOCS/obj_models/real_test

tianbartek, 2022-06-01:
Hi Wenbo, 

Thank you for sharing the code on GitHub. I followed your steps to install/download everything. And when I ran the code (run_nocs.py), it remind me that: "Could not find file NOCS/object_models/real_test/*.mtl".  When I checked the NOCS dataset website, I didn't find *.mtl in "object_model". 

Could you please let me know where I can get *.mtl files? 

Many thanks

tianbartek, 2022-06-06:
Hi, thanks for your reply. But it reminds me cannot find *.mtl, and stop running. What should I do?
Looking forward to your reply. Thanks again.

wenbowen123, 2022-06-06:
That's strange. Can you paste the complete error output here?

tianbartek, 2022-06-09:
Hi WenBo, 

Thank you for your reminder. I made the wrong data directories. Now, it works. Many thanks. 

Best wishes

ZhouKaichen, 2022-06-10:
> Hi WenBo,
> 
> Thank you for your reminder. I made the wrong data directories. Now, it works. Many thanks.
> 
> Best wishes

hello,can you tell me how to solve this errors?I meet too......