wenbowen123/BundleTrack #48: A possible solution to "findNN too few match"

baichuan05, 2022-06-01:
I am running on an older GPU card, GTX Titan X (Maxwell cards). Running it for the YCB object.

I was having a similar issue of "findNN too few match" as previous issues: https://github.com/wenbowen123/BundleTrack/issues/9 https://github.com/wenbowen123/BundleTrack/issues/15

I tried to compile OpenCV from the source but it didn't help. 
**Then, I tried to add `-gencode arch=compute_52,code=sm_52` in CMakeList.txt** https://github.com/wenbowen123/BundleTrack/blob/3df16853c745e7f216ea45df386a57fff0fc9c39/CMakeLists.txt#L7
It works now after rebuilding BundleTrack, I tried for ycbineoat/bleach0. The images in color_viz show the mask and also the yellow points for every frame.

I imagine it may also run on a newer GPU (Ampere), if you add something like `sm_86, compute_86`.

Last, Bowen, do you have visualization code to plot bounding boxes like your gifs that can be released?
Thanks!

wenbowen123, 2022-06-01:
Thanks for bringing it up, Baichuan! Yes that could work.

Here is the code to plot bbox
```
def draw_bbox(K, img, ob_in_cam, bbox, color_id, linewidth=2):
 def search_fit(points):
   '''
   @points: (N,3)
   '''
   min_x = min(points[:, 0])
   max_x = max(points[:, 0])
   min_y = min(points[:, 1])
   max_y = max(points[:, 1])
   min_z = min(points[:, 2])
   max_z = max(points[:, 2])
   return [min_x, max_x, min_y, max_y, min_z, max_z]
 
 def build_frame(min_x, max_x, min_y, max_y, min_z, max_z):
   bbox = []
   for i in np.arange(min_x, max_x, 1.0):
           bbox.append([i, min_y, min_z])
   for i in np.arange(min_x, max_x, 1.0):
           bbox.append([i, min_y, max_z])
   for i in np.arange(min_x, max_x, 1.0):
           bbox.append([i, max_y, min_z])
   for i in np.arange(min_x, max_x, 1.0):
           bbox.append([i, max_y, max_z])
 
   for i in np.arange(min_y, max_y, 1.0):
           bbox.append([min_x, i, min_z])
   for i in np.arange(min_y, max_y, 1.0):
           bbox.append([min_x, i, max_z])
   for i in np.arange(min_y, max_y, 1.0):
           bbox.append([max_x, i, min_z])
   for i in np.arange(min_y, max_y, 1.0):
           bbox.append([max_x, i, max_z])
 
   for i in np.arange(min_z, max_z, 1.0):
           bbox.append([min_x, min_y, i])
   for i in np.arange(min_z, max_z, 1.0):
           bbox.append([min_x, max_y, i])
   for i in np.arange(min_z, max_z, 1.0):
           bbox.append([max_x, min_y, i])
   for i in np.arange(min_z, max_z, 1.0):
           bbox.append([max_x, max_y, i])
   bbox = np.array(bbox)
   return bbox
 
 cam_cx = K[0,2]
 cam_cy = K[1,2]
 cam_fx = K[0,0]
 cam_fy = K[1,1]
 
 target_r = ob_in_cam[:3,:3]
 target_t = ob_in_cam[:3,3]*1000
 
 target = copy.deepcopy(bbox)
 limit = search_fit(target)
 bbox = build_frame(limit[0], limit[1], limit[2], limit[3], limit[4], limit[5])
 
 bbox = np.dot(bbox, target_r.T) + target_t
 
 color = np.array([[255, 69, 0], [124, 252, 0], [0, 238, 238], [238, 238, 0], [155, 48, 255], [0, 0, 238], [255, 131, 250], [189, 183, 107], [165, 42, 42], [0, 234, 0]])
vis = cv2.cvtColor(img, cv2.COLOR_BGR2RGB)
 for tg in bbox:
   y = int(tg[0] * cam_fx / tg[2] + cam_cx)
   x = int(tg[1] * cam_fy / tg[2] + cam_cy)
 
   if x - linewidth < 0 or x + linewidth > 479 or y - linewidth < 0 or y + linewidth > 639:
     continue
 
   for xxx in range(x-linewidth+1, x+linewidth):
     for yyy in range(y-linewidth+1, y+linewidth):
       vis[xxx][yyy] = color[color_id]
 
 vis = cv2.cvtColor(vis, cv2.COLOR_RGB2BGR)
 return vis

```

closed 2022-06-02 by baichuan05

baichuan05, 2022-06-13:
I can run it on RTX 3090 Ti.

Have a docker image from Nvidia-20.04-cuda11.2.2

Change `-gencode arch=compute_86,code=sm_86`

Compile OpenCV (4.5.2) with CUDA

Compile or apt install some other packages as listed in CMakeList

dennisushi, 2023-03-30:
I get 

```bash
nvcc fatal   : Unsupported gpu architecture 'compute_86'
CMake Error at cuda_compile_1_generated_cuda_ransac.cu.o.Release.cmake:220 (message):
  Error generating
  /home/dennisushi/repos/pose_estimation/BundleTrack/build/CMakeFiles/cuda_compile_1.dir/src/cuda/./cuda_compile_1_generated_cuda_ransac.cu.o

CMakeFiles/BundleTrack.dir/build.make:125: recipe for target 'CMakeFiles/cuda_compile_1.dir/src/cuda/cuda_compile_1_generated_cuda_ransac.cu.o' failed
```

@baichuan05 , how did you get around this?

wenbowen123, 2023-03-31:
We have a different [docker version for 3090 GPU](https://hub.docker.com/layers/wenbowen123/bundletrack/3090/images/sha256-55b844caf88344bce28ad21638977aeb5c621bd5d3b7479cc082cc625ebd389a?context=repo) if you want to try out.