wenbowen123/BundleTrack #49: Segmentation fault

cuiliangxian, 2022-06-12:
Hi，bowen，I met somes problem when setting up environmennt with your provided instructions. I check the file is not empty. can you please help me figure out this problem？
[image]

cuiliangxian, 2022-06-12:
the problem is solved, I missed to replace the path. But here comes another question. Did i forget some important files?
[image]

cynthia-you, 2022-06-29:
> the problem is solved, I missed to replace the path. But here comes another question. Did i forget some important files? [image]

I think you miss placed the obj_models path

wenbowen123, 2022-07-01:
What's your problem? The .mtl file missing error can be ignored. You should still be able to run.

cynthia-you, 2022-07-01:
> 

Thanks for you replay , the problem has been fixed. It was just a delay in communicating with another Docker port or a slow data read.

cynthia-you, 2022-07-01:
> What's your problem? The .mtl file missing error can be ignored. You should still be able to run.

Thanks for you replay , the problem has been fixed. It was just a delay in communicating with another Docker port or a slow data read.

closed 2022-07-01 by wenbowen123