wenbowen123/BundleTrack #50: A weird error when compiling

shanice-l, 2022-06-17:
Hi, bowen!

Thanks for your great job!

I'm trying to reproduce your results. But when compiling, I got a weird error

'''
[ 81%] Linking CXX shared library libBundleTrack.so
/usr/bin/ld: CMakeFiles/cuda_compile_1.dir/src/cuda/cuda_compile_1_generated_CUDAImageUtil.cu.o: relocation R_X86_64_PC32 against symbol `_ZN13CUDAImageUtil39adaptiveBilateralFilterIntensity_KernelEPfPKfS2_fffjj' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/BundleTrack.dir/build.make:459: libBundleTrack.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:134: CMakeFiles/BundleTrack.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
'''

This was happening when running

'''
mkdir build && cd build && cmake .. && make
'''

Any ideas of why this happens?

wenbowen123, 2022-07-01:
Hi, did you run this inside our provided docker?

shanice-l, 2022-07-01:
Thanks. Back then I don't know how to use docker. Now the problem is solved.

closed 2022-07-01 by shanice-l