wenbowen123/BundleTrack #51: ROS Integration?

cmower, 2022-06-20:
Hi @wenbowen123, is this integrated with or is there plans to integrate with ROS?

robertokcanale, 2022-06-24:
I need it too! 

Is there any integration just with a simple/real camera?

wenbowen123, 2022-07-01:
@robertokcanale @cmower 
For now, it does not support ROS yet. We may update it some time later. Stay tuned.

closed 2022-07-01 by wenbowen123

raghavauppuluri13, 2022-07-17:
@cmower @robertokcanale I added code to integrate the realtime tracking with ROS melodic in [my fork](https://github.com/raghavauppuluri13/BundleTrack). I wasn't able to fully test it because I have an unsupported GPU (3080), but feel free to try it out after adapting code in the `src/tracker_node.cpp` file.

kurshakuz, 2022-10-18:
@raghavauppuluri13 were you able to test it out?