wenbowen123/BundleTrack #57: Some questions about comparing the results of the experiment.

BoBoINVIDICUS, 2022-10-17:
Hello wenbowen, I would like to know how the comparative experiment results in your paper are obtained? Have you run every other algorithm on a public dataset, or can you find the results of other algorithms on a website? How is each performance metric obtained?
[image]

wenbowen123, 2022-10-18:
For most comparison methods, we directly adopt them from the [6-pack paper ](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwjxktny8Or6AhVDIzQIHU1MBZcQFnoECBAQAQ&url=https%3A%2F%2Farxiv.org%2Fabs%2F1910.10750&usg=AOvVaw2Dugn_O06VYLpr3Z1M_B07)

For others we run the open-sourced code on the public datasets. Please check our paper for more details.

closed 2022-10-18 by wenbowen123

BoBoINVIDICUS, 2022-10-19:
> For most comparison methods, we directly adopt them from the [6-pack paper [...]
Thank you very much for your answer!