wenbowen123/BundleTrack #62: There's no folder called annotated_poses

BoBoINVIDICUS, 2022-11-13:
Hi, wenbo:
   When I  predicted on my own RGBD data, Some error messages appear, as shown in the following figure:
[image]
   But I followed your tutorial to create my own data set and didn't see the information about the folder "annotated_poses." Here's the structure of my data set:
[image]
Should I change some Settings in the file "run_ycbineoat.py"? Where to change?
Looking forward to your reply.

BoBoINVIDICUS, 2022-11-13:
When I  simply create the meaningless pose files with the same format as those in YCBInEOAT, 
[image]

Bear-kai, 2023-02-20:
The same problem! Have you solved it ?

Bear-kai, 2023-02-20:
> The same problem! Have you solved it ?

OK. I found my problem is due to different nameing style of files in rgb/ and depth/. Make sure they have exactly the same name. (eg. rgb/0001.png vs. depth/0001.png)

closed 2023-03-31 by wenbowen123