wenbowen123/BundleTrack #63: sm_86 architecture

neox395, 2022-11-16:
Hey Bowen,

thank you for your interesting research in the pose tracking regime ;)

I am currently trying to get your repository to run. Unfortunately hardware compability issues are the limiting factor.
The sm_86 architecture (in my case a NVIDIA 3090 TI GPU, with CUDA 11.8), is not supported. I tried to adapt the code within the bundletrack docker image, resulting in segmentation faults when dealing with the point cloud library.

Is it possible to
- release another docker image (wenbowen123/bundletrack:latest), for the sm_86 architecture?
- include the specific library versions in the README.md or CMakeLists.txt in case the previous point is not possible due to time constraints?

Best regards,
Gerhard

wenbowen123, 2023-03-31:
Hi, we have a different docker version that supports newer GPU such as 3090
https://hub.docker.com/layers/wenbowen123/bundletrack/3090/images/sha256-55b844caf88344bce28ad21638977aeb5c621bd5d3b7479cc082cc625ebd389a?context=repo

Neal2020GitHub, 2023-04-24:
> [](https://hub.docker.com/layers/wenbowen123/bundletrack/3090/images/sha256-55b844caf88344bce28ad21638977aeb5c621bd5d3b7479cc082cc625ebd389a?context=repo)

Hi Bowen, I tried to use this new docker to run on 3090. However, I met with "vtk" error when compiling. Specifically, in the "BundleTrack/src/Utils.cpp", there raised "vtkPolyLine.h: No such file or directory" error. Since I am not quite familiar with docker, I'm wondering how can I solve this problem. Look forward to your reply.

Neal2020GitHub, 2023-04-24:
> Hi, we have a different docker version that supports newer GPU such as 3090 https://hub.docker.com/layers/wenbowen123/bundletrack/3090/images/sha256-55b844caf88344bce28ad21638977aeb5c621bd5d3b7479cc082cc625ebd389a?context=repo

Hi Bowen, I tried to use this new docker to run on 3090. However, I met with "vtk" error when compiling. Specifically, in the "BundleTrack/src/Utils.cpp", there raised "vtkPolyLine.h: No such file or directory" error. Since I am not quite familiar with docker, I'm wondering how can I solve this problem. Look forward to your reply.

Neal2020GitHub, 2023-04-24:
> Hi, we have a different docker version that supports newer GPU such as 3090 https://hub.docker.com/layers/wenbowen123/bundletrack/3090/images/sha256-55b844caf88344bce28ad21638977aeb5c621bd5d3b7479cc082cc625ebd389a?context=repo

Hi Bowen, I tried to use this new docker to run on 3090. However, I met with "vtk" error when compiling. Specifically, in the "BundleTrack/src/Utils.cpp", there raised "vtkPolyLine.h: No such file or directory" error. Since I am not quite familiar with docker, I'm wondering how I can solve this problem. Look forward to your reply.

wenbowen123, 2023-05-03:
@Neal2020GitHub Thanks for reporting the bug. The code has been updated. Please try again.

Neal2020GitHub, 2023-05-03:
> @Neal2020GitHub Thanks for reporting the bug. The code has been updated. Please try again.

It works now. Thanks a lot.

closed 2023-05-03 by wenbowen123