wenbowen123/BundleTrack #65: A strange question

lgq-gpu, 2022-11-28:
First of all, thank you for your contribution, I encountered a strange problem here, the program automatically stops when it runs to the Connected to port 5555, I use a graphics card 2060, there is no new problem, please help me to analyze, the card has been here for a long time.
[image]

wenbowen123, 2023-03-31:
Hi, sorry for the late reply. Can you add some printing around [here](https://github.com/wenbowen123/BundleTrack/blob/690b4e24ddd16a69a8ac5078514a702a0c7cd414/src/app/bundle_track_nocs.cpp#L68) to see where it stops exactly?

lgq-gpu, 2023-04-09:
Thank you for your reply. I will try to output it. Thanks again

closed 2023-05-03 by wenbowen123