wenbowen123/BundleTrack #67: Ground truth pose of NOCS and your converted ones are different

royzhang12, 2023-02-24:
Hi @wenbowen123,
Thank you for your excellent work. I just have a question about the ground truth pose of the NOCS dataset. 

I have checked the gt pose in your converted ones (under gts/real_test_text) and the ones downloaded from the original NOCS dataset(under gts/real_test/result_real_test_scene_*_*.pkl). The translations of the two poses are the same while the rotations are different. Would you mind telling me why they are different? Are you using a different coordinates?

Best regards,
Ruo Zhang

Guptajakala, 2023-02-24:
Yes, it's a different coordinate system. I remember NOCS was using openGL camera coordinate, while we were using OpenCV convention (X-axis right, Y-axis down).

wenbowen123, 2023-02-24:
Yes, it's a different coordinate system. I remember NOCS was using openGL camera coordinate, while we were using OpenCV convention (X-axis right, Y-axis down).

royzhang12, 2023-03-03:
Thank you for the nice explanation. Also find that the pose in the original dataset has scaled rotations, which means the det(R) is not identity.

closed 2023-03-03 by royzhang12