wenbowen123/BundleTrack #69: Both the camera and object move

Neal2020GitHub, 2023-04-05:
Hi! What a nice work! I'm wondering how to use BundleTrack to handle the situation that both the camera and object move?

wenbowen123, 2023-04-06:
regardless of the movement of camera or object or both, BundleTrack will be able to track the object->camera pose. If you are interested in getting their poses in the world, then you need some camera localization module, to get cam->world pose, and then the object->world pose can also be inferred.

Neal2020GitHub, 2023-04-07:
Hi Bowen! Thanks for your reply! I was a bit confused about choosing camera or world as reference to compute pose change but now it seems both correct with reference to each coordinate.

closed 2023-04-11 by Neal2020GitHub