wenbowen123/BundleTrack #74: no correspondence can be found between frame for all data

nuaayxy, 2023-06-08:
New frame 1578966079884481894
zmq start waiting for reply
zmq got reply
finding corres between 1578966079884481894(id=1) and 1578966042136484471(id=0)
frame 1578966079884481894 and 1578966042136484471 findNN too few match, 1578966079884481894 status marked as FAILcolor file: /home/xye/BundleTrack/YCBInEOAT/bleach0//rgb/1578966079932383160.png

New frame 1578966079932383160
zmq start waiting for reply
zmq got reply
finding corres between 1578966079932383160(id=1) and 1578966042136484471(id=0)
frame 1578966079932383160 and 1578966042136484471 findNN too few match, 1578966079932383160 status marked as FAILcolor file: /home/xye/BundleTrack/YCBInEOAT/bleach0//rgb/1578966079986252670.png

New frame 1578966079986252670
zmq start waiting for reply
zmq got reply
finding corres between 1578966079986252670(id=1) and 1578966042136484471(id=0)
frame 1578966079986252670 and 1578966042136484471 findNN too few match, 1578966079986252670 status marked as FAILcolor file: /home/xye/BundleTrack/YCBInEOAT/bleach0//rgb/1578966080035146972.png

New frame 1578966080035146972
zmq start waiting for reply
zmq got reply
finding corres between 1578966080035146972(id=1) and 1578966042136484471(id=0)
frame 1578966080035146972 and 1578966042136484471 findNN too few match, 1578966080035146972 status marked as FAILcolor file: /home/xye/BundleTrack/YCBInEOAT/bleach0//rgb/1578966080086818728.png

WASEDACHENZIJIAN, 2023-07-31:
@nuaayxy Hi，I also met the same problem as you, did you solve it?

closed 2023-07-31 by wenbowen123

nuaayxy, 2023-08-21:
I changed cmakelist but still some issue it only able to find 1 pair for every frame. 

New frame 1578966064570185448
zmq start waiting for reply
zmq got reply
finding corres between 1578966064570185448(id=394) and 1578966064501598691(id=393)
start multi pair ransac GPU, pairs#=1
total keyframes=12, already chosen _local_frames=1, want to select 15
#_local_frames=13
1578966042136484471 1578966050469548330 1578966050717028932 1578966050872408986 1578966051025319618 1578966059410082444 1578966059681959864 1578966059986456531 1578966060337009669 1578966060742621737 1578966061177809533 1578966061779361025 1578966064570185448 
finding corres between 1578966064570185448(id=394) and 1578966042136484471(id=0)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966064570185448(id=394) and 1578966050469548330(id=147)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966064570185448(id=394) and 1578966050717028932(id=151)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966064570185448(id=394) and 1578966050872408986(id=154)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966064570185448(id=394) and 1578966051025319618(id=157)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966064570185448(id=394) and 1578966059410082444(id=305)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966064570185448(id=394) and 1578966059681959864(id=310)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966064570185448(id=394) and 1578966059986456531(id=314)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966064570185448(id=394) and 1578966060337009669(id=320)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966064570185448(id=394) and 1578966060742621737(id=327)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966064570185448(id=394) and 1578966061177809533(id=335)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966064570185448(id=394) and 1578966061779361025(id=346)
start multi pair ransac GPU, pairs#=1
OptimizerGPU begin, global_corres#=9341
global_corres=9341
maxNumResiduals / maxNumberOfImages = 383741 / 13 = 29518
m_maxNumberOfImages*m_maxCorrPerImage = 13 x 1804 = 23452
m_solver->solve Time difference = 18.207[ms]
color file: /home/xye/BundleTrack/YCBInEOAT/bleach0//rgb/1578966064628422838.png