wenbowen123/BundleTrack #77: About feature detection FAIL

WASEDACHENZIJIAN, 2023-07-28:
[image]

WASEDACHENZIJIAN, 2023-07-28:
@wenbowen123 Hi，I followed your steps and tried the NOCS and YCBInEOAT datasets respectively, but there was a problem of feature detection fail. Do you know where the problem is?

wenbowen123, 2023-07-28:
Can you make sure your depth image is in the right format? You can enable higher debug level to dump the point cloud and viz them.

WASEDACHENZIJIAN, 2023-07-29:
[image]
[image]

The depth file is obtained from bleach0. There are 663 images, I think it should be right.

wenbowen123, 2023-07-31:
https://github.com/wenbowen123/BundleTrack/blob/5f87ddf7133f2f86bdae6a08c926ed80a70003ec/src/Bundler.cpp#L118
Before finding feature correspondence, can you check if the depth image values are normal? I remember there were some similar issues previously and it was because of the depth are all zeros.

WASEDACHENZIJIAN, 2023-08-01:
@wenbowen123 Hi, thank you for your help, I have already solve the problem about feature detection FAIL. I update CUDA version from 9.0 to 11.3. 

But now, there are some other problems, about FindNN too few match.
I refer to previous answers, but it doesn't seem to solve my problem.

[image]

About the depth image values，like this：
[image]

My server is two 3090, CUDA version is 11.3.

I also modified 

add_definitions(-DNO_OPENCV_CUDA=1),

 but it has no effect.

WASEDACHENZIJIAN, 2023-08-01:
@wenbowen123 

According to the current results, it seems that the values of ptA.z and ptB.z are too large, causing all of them to be filtered out. I don't know why.

wenbowen123, 2023-08-03:
can you check if the depth is like this at the beginning (after read), or is it changed to such somewhere?

WASEDACHENZIJIAN, 2023-08-05:
@wenbowen123  Hi, 
[image]
 In this function：SiftManager::pruneMatches (FeatureManager.cpp)
[image]
We can find that ptA.z and ptB.z are larger than 0.1.
But they will still be judged as less than 0.1, which is very strange.

WASEDACHENZIJIAN, 2023-08-05:
[image]
[image]
The depth value becomes 0 at this place

WASEDACHENZIJIAN, 2023-08-05:
[image]
[image]

wenbowen123, 2023-08-05:
your printing format is not right. Can you replace your `%d` by `%f` and retry?

WASEDACHENZIJIAN, 2023-08-06:
Thanks for your correction! The result is as follows：
[image]
[image]

wenbowen123, 2023-08-12:
it seems like the depth is already 0 at the line you printed. Can you trace back where it happened like I said?
>can you check if the depth is like this at the beginning (after read), or is it changed to such somewhere?

WASEDACHENZIJIAN, 2023-08-13:
@wenbowen123 

I have solved all the problems. Thanks for your help ！

My GPU is 3090, 

I would like to offer a possible solution to this problem:

Modify the CMakeLists.txt file:

From：
`set(CUDA_NVCC_FLAGS ${CUDA_NVCC_FLAGS} -Xcompiler;-fPIC;-gencode arch=compute_61,code=sm_61;-gencode arch=compute_75,code=sm_75;-O3;-std=c++11;-use_fast_math;--default-stream per-thread)`

To：
`set(CUDA_NVCC_FLAGS ${CUDA_NVCC_FLAGS} -Xcompiler;-fPIC;-gencode arch=compute_61,code=sm_61;-gencode arch=compute_75,code=sm_75;-gencode arch=compute_86,code=sm_86;-O3;-std=c++11;-use_fast_math;--default-stream per-thread)`

closed 2023-08-13 by WASEDACHENZIJIAN