wenbowen123/BundleTrack #91: feature detection failed on NVIDIA 4090

jhchong0405, 2024-03-11:
Hi @wenbowen123,

I am receiving "frame marked as FAIL since feature detection failed, ERROR." when I run BundleTrack.

After reading the issues, I have made some modifications to adapt to the 4090 gpu. 

Here are the modifications I made:
1. In the CMakeList.txt file, I made changes to the following line of code:

'''
set(CUDA_NVCC_FLAGS ${CUDA_NVCC_FLAGS} -Xcompiler;-fPIC;-gencode arch=compute_61,code=sm_61;-gencode arch=compute_75,code=sm_75;-gencode arch=compute_86,code=sm_86;-gencode arch=compute_89,code=sm_89;-O3;-std=c++11;-use_fast_math;--default-stream per-thread)
'''

^ added "-gencode arch=compute_89,code=sm_89" to suit 4090.

2. In the Docker environment, I updated CUDA to version 11.8. When building the app, I found that OpenCV needed to be compatible with the CUDA version, so I installed the latest version of OpenCV (4.9.0). This allowed us to run BundleTrack, but the pose tracking model was unable to detect features from our images. The issue showed the error message "frame marked as FAIL since feature detection failed, ERROR."

I have checked that my depth images are in the right format (uint16 & in mm unit). I have also read #77 and ensure that our CUDA version match with 4090. 

Can you please advise on whether our modification is correct and can you please guide us on how to solve this problem? Thanks!

wenbowen123, 2024-03-12:
is the error `"frame marked as FAIL since feature detection failed, ERROR." ` happenning from the first frame, or from some middle point of the video?

wenbowen123, 2024-03-12:
Also I'd encourage to check out our more recent work [BundleSDF](https://github.com/NVlabs/BundleSDF) which has some improvements on the results.

jhchong0405, 2024-03-14:
> Also I'd encourage to check out our more recent work [BundleSDF](https://github.com/NVlabs/BundleSDF) which has some improvements on the results.

Previously, we have tried running the code on another laptop with a NVIDIA 4060 gpu and a compute capability of 8.6 (same as 3090) and it was running fine. Therefore, we don't think it is an issue with the model performance. It seems like it is an issue with the environment.

jhchong0405, 2024-03-14:
Hi, we have solved our problem by recompiling OpenCV. We printed the exception below and found out that it was because our OpenCV didn't compile with cuda enabled. 
https://github.com/wenbowen123/BundleTrack/blob/5f87ddf7133f2f86bdae6a08c926ed80a70003ec/src/Bundler.cpp#L110

```
Error:

Exception caught: OpenCV(4.9.0-dev) /opencv_build/opencv/modules/core/include/opencv2/core/private.cuda.hpp:106: error: (-216:No CUDA support) The library is compiled without CUDA support in function 'throw_no_cuda’

frame marked as FAIL since feature detection failed, ERROR
```

Then, we solved the problem by following tutorial to recompile OpenCV and it is working now.

https://www.jeremymorgan.com/tutorials/linux/how-to-opencv-cuda-arch-linux/

closed 2024-03-14 by jhchong0405