wenbowen123/BundleTrack #94: Aborted (core dumped) on 3070 Ti with wenbowen123/bundletrack:3090

kirilllzaitsev, 2024-09-14:
Hi, I'm trying to run the code on a machine with the following `nvidia-smi` spec:
```
Sat Sep 14 23:25:55 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.183.01             Driver Version: 535.183.01   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 3070 Ti    Off | 00000000:01:00.0  On |                  N/A |
| N/A   42C    P3              22W / 115W |    328MiB /  8192MiB |     21%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
```

Using the image `wenbowen123/bundletrack:3090`, I'm facing the error:
```
(py38) root@host:/BundleTrack# python scripts/run_nocs.py --nocs_dir /data/nocs --scene_id 1 --port 5555 --model_name can_arizona_tea_norm
/BundleTrack/scripts/../build/bundle_track_nocs /tmp/config_model_can_arizona_tea_norm_scene1.yml
K
591.013       0 322.525
      0 590.168 244.111
      0       0       1

terminate called after throwing an instance of 'std::invalid_argument'
  what():  stof
Aborted (core dumped)
```

I've found some issues related to `Aborted (core dumped)` yet the resolution seemed to be either `3090` tag of the docker image or an older GPU. I'd like to ask if an older GPU is the only option having `3070 Ti`?

kirilllzaitsev, 2024-09-15:
Extending `CUDA_NVCC_FLAGS` with Ampere architecture:
`set(CUDA_NVCC_FLAGS ${CUDA_NVCC_FLAGS} -Xcompiler;-fPIC;-gencode arch=compute_61,code=sm_61;-gencode arch=compute_75,code=sm_75;-gencode arch=compute_86,code=sm_86;-O3;-std=c++11;-use_fast_math;--default-stream per-thread)`

 results in the same problem

kirilllzaitsev, 2024-09-15:
Interestingly, the same setup works for YCBInEOAT:
```
New frame 1578966080254803408
zmq start waiting for reply
zmq got reply
finding corres between 1578966080254803408(id=661) and 1578966080190493266(id=660)
start multi pair ransac GPU, pairs#=1
total keyframes=17, already chosen _local_frames=1, want to select 15
#_local_frames=15
1578966042136484471 1578966051025319618 1578966059410082444 1578966059627524281 1578966060160583495 1578966060636313767 1578966061070982573 1578966061676847697 1578966067725148647 1578966068006247984 1578966068982284853 1578966069219372833 1578966080035146972 1578966080190493266 1578966080254803408 
finding corres between 1578966080254803408(id=661) and 1578966042136484471(id=0)
finding corres between 1578966080254803408(id=661) and 1578966051025319618(id=157)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966080254803408(id=661) and 1578966059410082444(id=305)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966080254803408(id=661) and 1578966059627524281(id=309)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966080254803408(id=661) and 1578966060160583495(id=317)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966080254803408(id=661) and 1578966060636313767(id=325)
finding corres between 1578966080254803408(id=661) and 1578966061070982573(id=333)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966080254803408(id=661) and 1578966061676847697(id=344)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966080254803408(id=661) and 1578966067725148647(id=449)
finding corres between 1578966080254803408(id=661) and 1578966068006247984(id=454)
finding corres between 1578966080254803408(id=661) and 1578966068982284853(id=470)
start multi pair ransac GPU, pairs#=1
finding corres between 1578966080254803408(id=661) and 1578966069219372833(id=474)
finding corres between 1578966080254803408(id=661) and 1578966080035146972(id=657)
start multi pair ransac GPU, pairs#=1
OptimizerGPU begin, global_corres#=7532
global_corres=7532
maxNumResiduals / maxNumberOfImages = 511532 / 15 = 34102
m_maxNumberOfImages*m_maxCorrPerImage = 15 x 1572 = 23580
m_solver->solve Time difference = 20.676[ms]
color file: /data/ycbineoat/bleach0/rgb/1578966080320583789.png
```

Though resulting in an error before completion:
```
New frame 1578966080320583789
Frame 1578966080320583789 cloud is empty, marked FAIL, roi WxH=-9999.000000x-9999.000000
```

wenbowen123, 2024-09-16:
I guess it's related to the CUDA/GPU stuff. Btw, you can also check the follow up work [BundleSDF](https://github.com/NVlabs/BundleSDF)

closed 2024-09-27 by kirilllzaitsev